hanruihua/ir-sim #12: 有关多智能体强化学习收敛时间和难度的问题

sisijunn, 2024-10-29:
您好，我使用MADDPG和MATD3 在您的雷达小车环境中进行训练，但是基本上20k-30k的训练轮数后奖励曲线依旧无法收敛。我读过您的其他两篇文章和代码。您的代码在设置的时候默认训练轮次为250，请问您在训练的时候大概需要多少轮奖励曲线可以收敛？以及您是否用过其他的算法（MADDPG和MATD3）试验过，是不是这类算法就是比较难收敛所以您才选择用MAPPO

sisijunn, 2024-10-29:
[image]
这个是我的奖励曲线，我的奖励函数是到达目标点给5，碰撞给-5，每向目标点靠近给正向奖励，反之给负向奖励

sisijunn, 2024-10-29:
这个是我的奖励曲线，我的奖励函数是到达目标点给5，碰撞给-5，每向目标点靠近给正向奖励，反之给负向奖励
[image]

hanruihua, 2024-10-29:
你好

对于多智能体的奖励函数收敛问题，主要是和reward的计算方式也有关系。 因为对于多智能体而言，每次任务的失败是某两个angent 碰撞，成功是没有任何碰撞的到达目的地。 所以单纯的用reward的总和收敛来代表是否训练完成不够准确 （当然也有可能有更好的reward 计算方式)。 我目前没有测试过其他的算法。 我的代码里没有根据reward来判断收敛，是通过直接另开一个线程，测试当前model在100个case下计算成功率来判断是否训练完成的，供你参考。

sisijunn, 2024-10-30:
请问在您的实验中训练大概多少轮次成功率达到比较高的水平？

hanruihua, 2024-10-30:
4个机器人大概200 epoch左右， 在此基础上继续用10个机器人训练大概 1000 epoch 会达到比较高的水平。

sisijunn, 2024-10-30:
感谢您的回答，我这几天试过调整了一下学习率和奖励等超参数效果好一些了。

closed 2024-11-09 by hanruihua