hanruihua/ir-sim #14: Keyword argument static conflicts with default argument in ObjectStatic init

reiniscimurs, 2025-01-27:
Hi,

I was trying to create static obstacles with the yaml file and pass in an argument `static: True` (https://ir-sim.readthedocs.io/en/latest/yaml_config/configuration.html#static) but this seemingly causes an error when creating a static object since it also passes in a keyword argument that is already a default argument in the ObjectStatic init method https://github.com/hanruihua/ir-sim/blob/main/irsim/world/obstacles/obstacle_static.py#L8

My proposal is to either pop the `static` keyword in object factory (For instance here https://github.com/hanruihua/ir-sim/blob/main/irsim/world/object_factory.py#L142) or to remove the default `static = True` and add the keyword argument if `kinematics_name is None`

Additionally, what I am trying to achieve is to create static obstacles alongside dynamic ones. This causes an issue that for the static obstacles kinematics are still expected and I get a call to `"kinematics not implemented"`: https://github.com/hanruihua/ir-sim/blob/main/irsim/world/object_base.py#L1343

I would expect that at this point if there would be additional elif stattement  `elif self.kinematics == "static"` the model would return the `np.zeros((2, 1))` without the print statement.

You should be able to reproduce these issues with:
```yaml
obstacle:
  - number: 3
    kinematics: {name: 'omni'}
    distribution: {name: 'random', range_low: [0, 0, -3.14], range_high: [10, 10, 3.14]}
    behavior: {name: 'rvo', wander: True, range_low: [0, 0, -3.14], range_high: [10, 10, 3.14], vxmax: 0.2, vymax: 0.2, factor: 1.0}
    vel_max: [0.2, 0.2]
    vel_min: [-0.2, -0.2]
    shape:
      - {name: 'circle', radius: 1.0, random_shape: True}
      - {name: 'polygon', random_shape: true, avg_radius_range: [0.5, 1.0], irregularity_range: [0, 0.4], spikeyness_range: [0, 0.4], num_vertices_range: [4, 6]}

  - shape: { name: 'linestring', vertices: [ [ 0, 0 ], [ 10, 0 ], [ 10, 10 ], [ 0, 10 ],[ 0, 0 ]  ] }  # vertices
    static: True #remove this to see kinematics not implemented issue
    state: [ 0, 0, 0 ]
```

Please let me know if I am missing something here or there are any other ways to create static + dynamic obstacles together. Thanks!

hanruihua, 2025-01-28:
Hi, thank you for pointing this out. 

I have resolved this error by removing the default `static = True` in _object_factory.py_. Additionally, the print statement of  `"kinematics not implemented"  ` for static obstacles has also been removed. Please try running your code using the latest main branch.

Your yaml file is a right way to create static+dynamic obstacles. For each group of obstacles ( i.e. start with ` -` ), setting the `static=True` or  omitting the `kinematics `parameter will make the object static. 

A bug-fixed version will be realeased shortly.

reiniscimurs, 2025-01-28:
Thanks for the prompt response!

closed 2025-01-28 by reiniscimurs