hanruihua/ir-sim #15: Feature request for setting IDs in random_obstacle_position()

reiniscimurs, 2025-01-31:
Hi,

I am using the ir-sim to randomize the environment on every restart. There is some functionality that really helps with that, like setting goal and state functions. Especially useful is the `random_obstacle_position()` that randomizes the positions of the obstacles. However, there are some obstacles that I would like to remain in constant position throughout the execution. Currently the `random_obstacle_position()` does not take in arguments that would help to set position for some obstacles but not others. My proposal is to pass in a list of IDs of obstacles for which to randomize the position. I think this could be quite useful to others as well. Locally, I have implemented it as follows:

```python
    def random_obstacle_position(
        self, range_low=[0, 0, -3.14], range_high=[10, 10, 3.14], ids = []
    ):
        """
        Random obstacle positions in the environment.

        Args:
            range_low (list [x, y, theta]): Lower bound of the random range for the obstacle states. Default is [0, 0, -3.14].
            range_high (list [x, y, theta]): Upper bound of the random range for the obstacle states. Default is [10, 10, 3.14].
            ids (list): A list of IDs of objects for which to set random positions
        """
        if not len(ids):
            ids = [obs._id for obs in self.obstacle_list]
        if isinstance(range_low, list):
            range_low = np.c_[range_low]

        if isinstance(range_high, list):
            range_high = np.c_[range_high]

        random_states = np.random.uniform(
            range_low, range_high, (3, self.obstacle_number)
        )
        for i, obj in enumerate(self.obstacle_list):
            if obj._id in ids:
                obj.set_state(random_states[:, i].reshape(3, 1), init=True)

        self._env_plot.clear_components("all", self.obstacle_list)
        self._env_plot.draw_components("all", self.obstacle_list)
```

hanruihua, 2025-01-31:
Hi,

Thank you for taking the time to submit this feature request! Your suggestion and implementation are quite useful. 

If you're interested in contributing, please feel free to submit a pull request with your proposed changes.

A small suggestion: Consider using `obs.id` instead of `obs._id` for conciseness.

reiniscimurs, 2025-02-02:
I think that is a good idea. I opened a PR here: https://github.com/hanruihua/ir-sim/pull/16
Let me know if any additional details should be added there.

hanruihua, 2025-02-03:
Thanks for your contribution.

closed 2025-02-04 by reiniscimurs