hanruihua/ir-sim #17: Issues with lag

Adithya1976, 2025-02-12:
Thank you so much for this repository. It was a life-saver for someone who doesn't have resources to use 3D simulators like gazebo. I have noticed that as I increase the number of robots, the lag increases heavily. The lag is not much of an issue to me. However, I'm unsure if the kinematics of the cars are still accurate. I might be wrong but I'm assuming there's a fixed step time which can be configured and every frame that is displayed is not actually showing the robots position at the current time but its showing the robots position after (number of steps)*(step_time). Is that the case? Secondly, I noticed that all the entities in the environment get updated sequentially and not parallely. Could we address this by using libraries such as asyncio to perform parallel updates?

hanruihua, 2025-02-12:
Hi, thank you for your attention to my work.

- Regarding the lag issue, please don't worry about accuracy. Currently, the visualization (`env.render()`) and the robot state update (`env.step()`) are performed synchronously at each time step. Here, `step_time` represents the frequency of state updates, while `sample_time` determines the frequency of environment rendering. The `interval` refers to the elapsed time for frame rendering, which only affects animation visualization—not the simulation results—since the frame always shows the updated robot positions. The purpose of setting up `sample_time` is to reduce the computational cost associated with the matplotlib render; you can think of it as displaying the updated result at regular intervals.

- I don't know your specific configuration, but if you are running multiple robots equipped with lidar, heavy lag might occur due to collision checking in the lidar and the matplotlib rendering. To reduce lag, consider lowering the number of laser beams or adjusting the `sample_time`.

- I appreciate your advise. I am also considering improving the running speed by incorporating parallel computing and asynchronous modes. However, in this simulator, since each agent performs an atomic task, these changes don't lead to a significant speedup at the moment. I'll continue to prioritize this issue and look forward to its future resolution.

Adithya1976, 2025-02-13:
Thank you for addressing my issues. If time permits, I will try adding the asynchronous functionality to help with my project and contribute to this repo.

closed 2025-02-13 by Adithya1976