hanruihua/ir-sim #26: Question - more than one goal for the robot?

ljaniec, 2025-03-13:
Hello,

It is a really nice project, I am enjoying its simplicity. I have a question - is it possible to set up a more than one goal triplet `(x, y, theta)` for the robot? 

- If yes - do you have some example YAML file and world for that?
- If no - could you write for me what parts of the code should be probably modified to allow this?

Best,
Łukasz Janiec

hanruihua, 2025-03-13:
Hi,

I have added support for multiple goal setting #27. You can try the updated main branch by running the example 02robot_world to see the results. One specific setting to use is

```
  goal: 
    - [[9, 9, 0], [2, 4, 1], [3, 3, 2]]

```

in yaml file.

ljaniec, 2025-03-13:
That was fast! I tested it, and it worked well with this example. Thank you.

Additional note:

Previously at the start of trying the examles out, I had some errors regarding the proper requirements' installation, especially matplotlib and numpy - I saw some errors and warnings about erroneous versions 2.x and 1.y (even though I had `numpy==2.2.3` version, the one used there as well). 
It would help others to have the `requirements.txt` file. At last, separate installation of everything in the `venv` was enough, but I had to install the packages by hand. This is the result of `pip freeze` in the `venv` inside the repository:

```bash
contourpy==1.3.1
cycler==0.12.1
fonttools==4.56.0
imageio==2.37.0
-e git+https://github.com/hanruihua/ir-sim.git@31577184c9612382d682e82508724bdb698f44f8#egg=ir_sim
kiwisolver==1.4.8
loguru==0.7.3
matplotlib==3.9.4
numpy==2.2.3
packaging==24.2
pillow==11.1.0
pyparsing==3.2.1
python-dateutil==2.9.0.post0
PyYAML==6.0.2
shapely==2.0.7
six==1.17.0
tabulate==0.9.0
```

hanruihua, 2025-03-13:
Hi  

I appreciate your advice, and I've created the requirements.txt file for quick installation of dependencies.

closed 2025-03-22 by hanruihua