hanruihua/ir-sim #39: how to get sensor data in custom behaviour?

aly-pr, 2025-03-24:
Im trying to add a custom behaviour and use the lidar sensor data to help guide. Is this possible?

hanruihua, 2025-03-24:
Hi, you can use the function `object.get_lidar_scan()` to obtain the lidar sensor data. For example, in `custom_behavior_methods.py`:

```
@register_behavior("diff", "dash_custom")
def beh_diff_dash(ego_object, external_objects=[], **kwargs):

    print("This is a custom behavior example for differential drive with dash2")

    scan_data = ego_object.get_lidar_scan()  # get lidar sensor data
    scan_data_external = external_objects[0].get_lidar_scan()

    state = ego_object.state
    goal = ego_object.goal
    goal_threshold = ego_object.goal_threshold
    _, max_vel = ego_object.get_vel_range()
    angle_tolerance = kwargs.get("angle_tolerance", 0.1)

    behavior_vel = DiffDash2(state, goal, max_vel, goal_threshold, angle_tolerance)

    return behavior_vel
```

Additionally, please make sure that the lidar sensor is added in the yaml files for each robot.

hanruihua, 2025-03-24:
Hi, 

For the omni robot, the velocity returned should be cartesian `[vx, vy]`, `vx`  is the velocity in x axis and `vy`  is the velocity in y axis. linear + angular are for diff robot. 

Thank you for pointing this out, I will improve the documentation to make it more clear.

hanruihua, 2025-03-24:
Not yet, but that's a good idea. I'll start the GitHub discussion.