hanruihua/ir-sim #70: Collision only for a single robot

reiniscimurs, 2025-06-13:
Hi,

Recently I noticed that sometimes when there is a collision between multiple robots, only a single robot will get a collision state and not the other robot that is collided with. As an example, here is a warning message for such a collision:

`WARNING  | robot_1 collided with robot_0 at state [4.7, 7.33, -2.09]`

My expectation is that the robot_0 would also detect a collision in such a case, since, realistically, it is a collision between both robots. One robot cannot have a collision while the other one does not. My expected output would be:

`WARNING  | robot_1 collided with robot_0 at state [4.7, 7.33, -2.09]`
`WARNING  | robot_0 collided with robot_1 at state [4.98, 7.06, -2.56]`

I traced the issue back to `possible_collision_objects` call, that it is seemingly too restrictive and does not detect some objects. To reduce the restriction I used increased distance in the query function:
`candidates_index = tree.query(self.geometry, predicate="dwithin", distance=0.1)`
This allows to detect collisions of both robots.

My guess is that the query function uses some sort of an abstraction for detecting overlaps and it can result that an overlap is detected from one robots geometry but not detected from the other one. Increasing this distance should allow to give better candidates for collision detection.

Let me know if there are other ways to avoid this issue or maybe there are other ideas to try out here?

hanruihua, 2025-06-14:
Hi,

Thank you for pointing this out.

I found that this issue may be caused by the order in which the object’s `check_status` and `step` functions are called. When there are multiple robots, each robot’s state is updated and its collision status is checked sequentially. However, for the first robot, it checks collisions with other robots whose states have not yet been updated, resulting in incorrect collision statuses.

To fix this, I’ve created PR #72, where the `check_status` function is invoked after all objects and the geometry tree have been updated. Please review the PR and let me know whether this fixes the collision status issue.

hanruihua, 2025-06-15:
Thanks for your testing. I will merge this PR.

closed 2025-07-02 by reiniscimurs